waveform80/rpi-lgpio #3: ModuleNotFoundError: No module named 'lgpio'

chetan1836, 2023-12-27:
When I try to import RPi.GPIO, I get the following error. 
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/username/.local/lib/python3.7/site-packages/RPi/GPIO/__init__.py", line 13, in <module>
    import lgpio
ModuleNotFoundError: No module named 'lgpio'`

I tried installing lgpio module, but it didn't fix the problem

epsiro, 2023-12-30:
I'm having the same issue on Linux 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux.

```
Traceback (most recent call last):
    import RPi.GPIO as GPIO
  File "/home/username/venv/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 13, in <module>
    import lgpio
ModuleNotFoundError: No module named 'lgpio'
```

epsiro, 2023-12-30:
`pip install lgpio` gave me lgpio-0.0.0.2, but I managed to get `lgpio-0.2.2.0` by following the instructions from http://abyz.me.uk/lg/download.html and using `pip install .` in the lg/PY_LGPIO directory (within a virtual environment).

This solved the issue for me.

Ed1ks, 2024-01-13:
> `pip install lgpio` gave me lgpio-0.0.0.2, but I managed to get `lgpio-0.2.2.0` by following the instructions from http://abyz.me.uk/lg/download.html and using `pip install .` in the lg/PY_LGPIO directory (within a virtual environment).
> 
> This solved the issue for me.

I have the same problem as TO, but this doesnt work for me:

[image]

ExSign, 2024-02-16:
I also ran into problems building by downloading from http://abyz.me.uk/lg/download.html

Turned out I had the same issue as https://github.com/gpiozero/gpiozero/issues/1038#issuecomment-1877541397, means my system-wide lgpio was Version: 0.2.2.0 and the virtual environment one was lgpio Version: 0.0.0.2. 

I copied over following files from /usr/lib/python3/dist-packages/ to /home/pi/OctoPrint/lib/python3.11/site-packages/

> lgpio-0.2.2.0.egg-info
> lgpio.py
> _lgpio.cpython-311-aarch64-linux-gnu.so

Now its working.

waveform80, 2024-04-09:
The problems with pip installing lgpio *should* now be resolved for Python versions 3.9 thru 3.12 on armhf and arm64 platforms (since [this PR for packaging](https://github.com/joan2937/lg/pull/25) was merged and I built some [manylinux wheels](https://pypi.org/project/lgpio/#files) for PyPI).

I'll close the issue for now as I think this is resolved, but please feel free to re-open it if you're still encountering issues (if you do, please post the output from pip showing how it's currently failing to install lgpio, and any information on your platform if possible -- one oversight I'm aware of is that I've not built manylinux wheels for amd64 yet).

closed 2024-04-09 by waveform80

bodiroga, 2024-08-15:
Hi @waveform80!

I'm trying to build a tiny docker image to run a script that controlls a GPIO pin but I'm stucked with the "no module named 'lgpio' error" and I don't know what to do. The steps I will show you are being executed in a Raspberry Pi 4 with Bookworm:

```
cat /etc/debian_version
12.6
```

```
uname -a
Linux iom2040-test.local 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```
```
docker version
Version:          27.0.3
```

And for the minimal reproduction steps:

- Run a python alpine container:

```
docker run -it --rm --privileged --entrypoint sh python:3.12.5-alpine3.20
```
- Install rpi-lgpio using pip

```
pip install rpi-lgpio==0.6

Collecting rpi-lgpio==0.6
  Downloading rpi_lgpio-0.6-py3-none-any.whl.metadata (2.5 kB)
Collecting lgpio>=0.1.0.1 (from rpi-lgpio==0.6)
  Downloading lgpio-0.2.2.0-cp312-cp312-linux_armv7l.whl.metadata (963 bytes)
Downloading rpi_lgpio-0.6-py3-none-any.whl (11 kB)
Downloading lgpio-0.2.2.0-cp312-cp312-linux_armv7l.whl (383 kB)
Installing collected packages: lgpio, rpi-lgpio
Successfully installed lgpio-0.2.2.0 rpi-lgpio-0.6
```
```
ls -lh /usr/local/lib/python3.12/site-packages/

total 928K   
-rw-r--r--    1 root     root         119 Aug  7 22:02 README.txt
drwxr-xr-x    4 root     root        4.0K Aug 15 13:16 RPi
drwxr-xr-x    2 root     root        4.0K Aug 15 13:16 __pycache__
drwxr-xr-x    1 root     root        4.0K Aug 15 13:15 _distutils_hack
-rwxr-xr-x    1 root     root      795.3K Aug 15 13:16 _lgpio.cpython-312-arm-linux-gnueabihf.so
-rw-r--r--    1 root     root         151 Aug  7 22:02 distutils-precedence.pth
drwxr-xr-x    2 root     root        4.0K Aug 15 13:16 lgpio-0.2.2.0.dist-info
-rw-r--r--    1 root     root       68.1K Aug 15 13:16 lgpio.py
drwxr-xr-x    1 root     root        4.0K Aug 15 13:15 pip
drwxr-xr-x    2 root     root        4.0K Aug  7 22:02 pip-24.2.dist-info
drwxr-xr-x    3 root     root        4.0K Aug  7 22:02 pkg_resources
drwxr-xr-x    2 root     root        4.0K Aug 15 13:16 rpi_lgpio-0.6.dist-info
drwxr-xr-x    8 root     root        4.0K Aug  7 22:02 setuptools
drwxr-xr-x    2 root     root        4.0K Aug  7 22:02 setuptools-72.1.0.dist-info
drwxr-xr-x    4 root     root        4.0K Aug  7 22:02 wheel
drwxr-xr-x    2 root     root        4.0K Aug  7 22:02 wheel-0.44.0.dist-info
```
Everything looks fine to my eyes, but if I open a Python terminal and try to import the library, this is the result:

```
/ # python
Python 3.12.5 (main, Aug  7 2024, 21:58:36) [GCC 13.2.1 20240309] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.12/site-packages/RPi/GPIO/__init__.py", line 14, in <module>
    import lgpio
  File "/usr/local/lib/python3.12/site-packages/lgpio.py", line 186, in <module>
    import _lgpio
ModuleNotFoundError: No module named '_lgpio'
```

I asume that I'm doing something wrong, but are you able to spot what could the cause of the error be? I've been reading as much as I can for the last 24 hours, but I'm blind.

Many thanks for taking the time to read my message and I hope that you can give me some clue 👼🏻 

Best regards!

bodiroga, 2024-08-15:
Hi again @waveform80!

I have decided to start from scrach using a fresh Raspberry Pi OS Lite (64bits) installation and I think that I have found what the problem could be: I was using a Raspberry Pi installation with a 64bits kernel but a 32bits userspace 🫤 That's why in the rpi-lgpio installation process the armv7l wheel is downloaded: "Downloading lgpio-0.2.2.0-cp312-cp312-linux_**armv7l**.whl (383 kB)"

What I have achieved until now (with the fresh uSD card):

- Run a python alpine container:

```
docker run -it --rm --privileged --entrypoint sh python:3.12.5-alpine3.20
```

- Add compilation tools:

```
apk add alpine-sdk swig
```

- Install rpi-gpio:

```
/ # pip install rpi-lgpio==0.6
Collecting rpi-lgpio==0.6
  Using cached rpi_lgpio-0.6-py3-none-any.whl.metadata (2.5 kB)
Collecting lgpio>=0.1.0.1 (from rpi-lgpio==0.6)
  Using cached lgpio-0.2.2.0.tar.gz (90 kB)
  Preparing metadata (setup.py) ... done
Using cached rpi_lgpio-0.6-py3-none-any.whl (11 kB)
Building wheels for collected packages: lgpio
  Building wheel for lgpio (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      running bdist_wheel
      running build
      running build_py
      running build_ext
      building '_lgpio' extension
      swigging lgpio.i to lgpio_wrap.c
      swig -python -o lgpio_wrap.c lgpio.i
      creating build
      creating build/temp.linux-aarch64-cpython-312
      gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O3 -Wall -fPIC -Isrc -I/usr/local/include/python3.12 -c lgpio_wrap.c -o build/temp.linux-aarch64-cpython-312/lgpio_wrap.o
      In file included from lgpio_wrap.c:3199:
      src/lgpio.h:34:10: fatal error: linux/gpio.h: No such file or directory
         34 | #include <linux/gpio.h>
            |          ^~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lgpio
  Running setup.py clean for lgpio
Failed to build lgpio
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (lgpio)
```

No one said it was going to be easy... 😢 

Anyway, thanks for your help and sorry for the false report!