waveform80/rpi-lgpio #4: Troubles reading gpio pin - could be my coding!

mcgoy, 2024-01-08:
Whenever I attempt to out right read a gpio pin, the pin is set to low and low is always returned to this script.  No idea why!  Scripts not using sys.argv, but with fixed pins seem to work ok.  I use an 'r' for argv[2] to just do the input on pin:

#!/usr/bin/python
import sys
import RPi.GPIO as GPIO
states=["OFF","ON"]
pin = int(sys.argv[1])
action = str(sys.argv[2])
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(pin,GPIO.OUT)
if action=='n': GPIO.output(pin,GPIO.HIGH)
if action=='f': GPIO.output(pin,GPIO.LOW)
x = GPIO.input(pin)
print (x)
print (states[x])

Thank you for the great package.  I'm sure this is something I am doing wrong.  This is uname -a:  Linux pecan 6.5.0-1008-raspi #11-Ubuntu SMP PREEMPT Wed Nov 22 16:17:39 UTC 2023 armv7l armv7l armv7l GNU/Linux  version of RPi.GPIO is 0.7.2  Python version:
root@pecan:/usr/local/bin#  /usr/bin/python --version
Python 3.11.6
 This is what I get when I run the script:
root@pecan:/usr/local/bin# mgp.py 17 n
1
ON
root@pecan:/usr/local/bin# gpio -g read 17
1
root@pecan:/usr/local/bin# mgp.py 17 r
0
OFF
root@pecan:/usr/local/bin# gpio -g read 17
0

I don't do the cleanup because I desire for the pin to remain in whatever state the script sets.  I use this to turn relays on and off. 
 The original, broken version of RPi.GPIO will work as expected with this script, but it breaks many other important things such as add_event_detect on ubuntu.  Everything with legacy RPi.GPIO still works on the traditional Pi Linux.  Thanks in advance for any help!  I have a sad work around using gpio to do the 'r' with subprocess. Would like to keep the code all in python.

waveform80, 2024-04-09:
This is quite strange -- I can't reproduce the issue on either Ubuntu jammy (22.04, with kernel version 5.15.0-1049-raspi), or noble (24.04 with kernel version 6.8.0-1002-raspi). Here's my results on noble on a Pi 5 (jammy, running on a Pi 4 is exactly the same):

```console
$ ./rpi-lgpio-4.py 17 n
1
ON
$ ./rpi-lgpio-4.py 17 r
1
ON
```

Can I just check exactly what platform you're on? From what you've included I'm guessing Ubuntu mantic (23.10) armhf (aka armv7l), but what model of Pi (or I suppose, given lgpio isn't Pi specific, what computer) are you using?

I'm assuming when you say you're using rpi-lgpio version 0.7.2 you've just querying RPi.GPIO.VERSION (which is actually a lie -- it outputs "0.7.2" as that's the version of the "real" RPi.GPIO it endeavours to be compatible with; the package's actual version is currently 0.4). If you run `apt policy python3-rpi-lgpio` it should tell you the version of the package, but it'll very like be 0.4-something as that's all I've released on the Ubuntu packaging so far.

mcgoy, 2024-04-10:
Greetings sir!

Here is the query:

***@***.***:/usr/local/bin# apt policy python3-rpi-lgpio
python3-rpi-lgpio:
  Installed: 0.4-0ubuntu1
  Candidate: 0.4-0ubuntu1
  Version table:
*** 0.4-0ubuntu1 500
        500 http://ports.ubuntu.com/ubuntu-ports mantic/universe armhf Packages
        100 /var/lib/dpkg/status
***@***.***:/usr/local/bin#

Which shows I’m running mantic:

***@***.***:/usr/local/bin# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 23.10
Release:        23.10
Codename:       mantic

Put all the updates on and still get same results.  My code:
***@***.***:/usr/local/bin# cat manipulate_gpio_pin.py.raspberry
#!/usr/bin/python
#ver 2020.10.28.01
import sys
import RPi.GPIO as GPIO
import time
pin = int(sys.argv[1])
action = str(sys.argv[2])
states=["OFF","ON"]
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(pin, GPIO.IN)
GPIO.setup(pin,GPIO.OUT)
if action=='n':
   GPIO.output(pin,GPIO.HIGH)
if action=='f':
   GPIO.output(pin,GPIO.LOW)
if action=='r':
   x = GPIO.input(pin)
   GPIO.output(pin,x)
x = GPIO.input(pin)
GPIO.output(pin,x)
print (states[x])

results:
***@***.***:/usr/local/bin# manipulate_gpio_pin.py.raspberry 17 n
ON
***@***.***:/usr/local/bin# manipulate_gpio_pin.py.raspberry 17 r
OFF
***@***.***:/usr/local/bin#

Sorry for my troubles.  Probably something I did.  Here’s my apt list:

  839  apt remove python3-rpi.gpio
  840  apt install python3-rpi-lgpio
  976  history | grep apt
1009  apt install libretech-lgpio
1010  apt install libretech-gpio
1011  apt-cache search lgpio
1013  apt install gpiod
1122  apt-cache search numpy
1123  apt-cache search numpy | grep numpy
1124  apt install python3-numpy python3-numpy-dev
1149  apt-cache search i2c | more
1150  apt install i2c-tools
1185  history | grep apt | grep gpio
1186  history | grep apt
1187  apt-cache search bme280
1188  apt install python3-bme280 bme280-doc
1221  apt list --installed
1222  apt list --installed | more
1223  apt list --installed | greo gpio
1224  apt list --installed | grep gpio
1230  apt list --installed | grep gpio
1234  apt update
1235  apt upgrade
1269  history | grep apt
1270  apt update
1271  apt updgrade
1272  apt upgrade
1280  apt install minicom
1345  apt-cache search mcp3008
1351  apt install unzip
1382  apt-cache search spidev
1383  apt install python3-spidev spi-tools
1446  history | grep apt
1519  apt update
1520  apt upgrade
1683  apt update
1684  apt upgrade
1686  apt update
1742  apt update
1743  apt upgrade
1761  apt update
1762  apt upgrade
1791  apt policy python3-rpi-lgpi
1792  apt install python3-rpi-lgpi
1793  apt policy python3-rpi-lgpio

Here is my pip:
***@***.***:/usr/local/bin# history | grep pip
1346  pip3 mcp3008
***@***.***:/usr/local/bin# uname -a
Linux pecan 6.5.0-1013-raspi #16-Ubuntu SMP PREEMPT Thu Mar 14 15:47:52 UTC 2024 armv7l armv7l armv7l GNU/Linux

This all running on a Pi3B (does not say plus on the board).
cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 000000007cfb5810
Model           : Raspberry Pi 3 Model B Rev 1.2

***@***.***:/usr/local/bin# cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev ***@***.******@***.***:/usr/local/bin#>

Purchased off e-bay in the middle of the pandemic.

Very sorry for the troubles.  If I need to reload, that’s no issue at all.  Reloading is all I know!

Kind regards,
David McGaughey

kami83, 2024-04-12:
Hi, i also got some problems with the GPIO pins. If i run this script on my Raspi 5 version 0.4-1~rpt1 500

I get this error for the pins 18,25 (BCM):
./testpin.py 25 f
Traceback (most recent call last):
  File "/home/testpin.py", line 9, in <module>
    GPIO.setup(pin,GPIO.OUT)
  File "/usr/local/lib/python3.11/dist-packages/RPi/GPIO/__init__.py", line 677, in setup
    initial = _check(lgpio.gpio_read(_chip, gpio))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 903, in gpio_read
    return _u2i(_lgpio._gpio_read(handle&0xffff, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO not allocated'

If use for example pin 12. It works fine.

BR kami

mcgoy, 2024-04-20:
Greetings!

I did a fresh load of ubuntu server Manic on a different Pi.  Your packages work perfectly on this Pi.  The only differences that I can see from this Pi and the one that doesn’t work are these:
Working Pi:  Raspberry Pi 3 Model B Plus Rev 1.3
Broken Pi:     Raspberry Pi 3 Model B Rev 1.2

The BCM2835 are also different revisions.
Working Pi:  a020d3
Broken Pi:    a22082

The software installs are pretty much the same.  I use a build sheet with instructions to do the installation of the software.  I haven’t gotten around to containers.

Very sorry to have troubled you!  You have a great package and I think you very much, I appreciate it!

Kind regards,
David McGaughey