waveform80/rpi-lgpio #5: Strange Exception message

ullix, 2024-02-18:
Using rpi-lgpio on Raspi 5 as replacement for old RPi.GPIO 0.7.1. All works well when the program runs, but when I stop it I get this strange Exception message:

Exception ignored in: <function PWM.__del__ at 0x7fff9a608360>
Traceback (most recent call last):
  File "/home/ullix/.local/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 178, in __del__
  File "/home/ullix/.local/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 202, in stop
AttributeError: 'NoneType' object has no attribute 'error'

It happens after all GPIO functions are shut down and cleanup() had been called. It seems it prints out only after the program has exited.

The line 202 in the stop function is the 'except' line:

        try:
            lgpio.tx_pwm(_chip, self._gpio, 0, 0)
        except lgpio.error:
            pass
        lgpio.gpio_write(_chip, self._gpio, 0)
        self._running = False

waveform80, 2024-04-09:
That's intriguing. Can I ask how you installed lgpio? My guess is via pip, in which case back in February you would've unfortunately obtained a very out of date version. A fresh pip install (or upgrade) of lgpio *should* remedy this now that I've added some up-to-date [manylinux wheels on PyPI](https://pypi.org/project/lgpio/#files) for lgpio

ullix, 2024-04-10:
Of course, I installed via pip. I haven't see this recently, so maybe (?) I had reinstalled and the problem was gone? Don't remember.

waveform80, 2024-04-10:
Okay, I'll close this for now on the assumption that an old version of lgpio was involved. Please feel free to re-open if you see it again.

closed 2024-04-10 by waveform80