waveform80/rpi-lgpio #6: For which Raspi versions can the rpi-lgpio module be used?

ullix, 2024-02-29:
I am using rpi-lgpio on a Raspi 5 and all works well. But I don't find a way to use it on a Raspi 3. It fails with message "No module named 'RPi'".

Does rpi-lgpio support any older versions of Raspi? Which ones from the Raspi 1,2,3,4, and Zeros?

waveform80, 2024-04-09:
rpi-lgpio should work on any model of the Pi that supports ["new-style revision codes"](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes), and that's only because I was [too lazy](https://github.com/waveform80/rpi-lgpio/blob/eacda9ba51133aa6c815b9f98f344ab5256895b2/RPi/GPIO/__init__.py#L426-L428) to implement support for the old-style ones. Specifically that means it *should* work out of the box on anything beyond the original model 1's (the 2, 3, 4, 5, and Zeros *should* all be fine).

You can also force rpi-lgpio to use a specific revision code via the [`RPI_LGPIO_REVISION` env-var](https://rpi-lgpio.readthedocs.io/en/release-0.4/differences.html#pi-revision) so technically it can be made to work on any board that has a compatible GPIO configuration to the revision code you specify.

On your Pi 3 what OS are you running, and how have you attempted to install rpi-lgpio?

ullix, 2024-04-10:
How would I know whether my Pi that supports "new-style revision codes"? 

I made a work-around by checking the Raspi version and loading either the old one or the new one. Not a nice solution, but at least working.

waveform80, 2024-04-10:
> How would I know whether my Pi that supports "new-style revision codes"?

```
$ grep "^Revision" /proc/cpuinfo
Revision        : c04170
```

If the revision code is four characters, e.g. 0004 then it's [old-style](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#old-style-revision-codes). This really only applies to the original model 1's (including the B+, A+, and CM1); everything since then uses the [new-style](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes) which is six characters as shown above. Even then, as noted, you can use rpi-lgpio on old-style boards, you just need to "fake" the revision code with `RPI_LGPIO_REVISION`.

The only case in which you should be seeing "No module named 'RPi'" is if the module is not available in the environment you're trying to import it from (whether that's the system environment or a venv).

commit a9c6ae22 (2024-04-12): Add more info on supported boards

Closes #6
closed 2024-04-12 by waveform80