waveform80/rpi-lgpio #14: difference in MP environment, GPIOs not shared anymore

Brun059, 2024-06-16:
lgpio is greate as it made event_detect to work (again), thanks and thanks https://raspberrypi.stackexchange.com/questions/147332/rpi-gpio-runtimeerror-failed-to-add-edge-detection for pointing me to it.

Yet, lgpio behaves differently in multiprocessing setup. The "old" GPIO allowed to use [and change] GPIO settings across processes while lgpio confines them to the respective running process. 

In other words, in GPIO I could setup all GPIOs in the main process, briefly flip one of them from output to input and back in a subprocess and read its value in both, the main and the subprocess. With lgpio I can only access a GPIO within and from the respective subprocess. I cannot read its value from both, the main and subprocess. Attempts to initialize them globally and change one from out to in within the subprocess results in  lgpio.errer: 'GPIO busy' 

(reason for doing this wired thing: I use the same PIN as output and briefly as input as well. This to control an external relay in a specific way manually and trough RPI, some diodes ensure that I'm not producing an electrical shortcut.

waveform80, 2024-07-28:
I won't be able to fix this given this is a fundamental difference in how gpio works with the gpiochip devices: pins are reserved by a process and a separate process cannot then access that pin until the first process releases it or dies.

That said, `multiprocessing` is a module that should be explicitly mentioned in the corresponding [documentation section](https://rpi-lgpio.readthedocs.io/en/latest/differences.html#simultaneous-access).

Treating this as a documentation enhancement, because I'm afraid the functionality is a case of "won't/can't fix"

andyfaff, 2025-06-04:
I'm not sure if this is relevant to this question.
I'm trying to `add_event_callback` on two pins. Each of the pins is being monitored in a separate `multiprocessing.Process`. However, the callbacks weren't being triggered. I was bashing my head against the wall until I found the following solution, *don't `import RPi.GPIO` at the top level, import it in the target function for the `Process`* 

This doesn't work:

```python
import RPi.GPIO as gpio

def proc1(ev):
    def callback(channel):
        print("P11 triggered")

    gpio.setmode(gpio.BOARD)
    gpio.setup(11, gpio.IN, pull_up_down=gpio.PUD_OFF)
    gpio.add_event_detect(11, gpio.RISING)
    gpio.add_event_callback(11, callback)

    while True:
        ev.wait(1.0)
        if ev.is_set():
            break
    gpio.cleanup()

def proc2(ev):
    def callback(channel):
        print("P13 triggered")

    gpio.setmode(gpio.BOARD)
    gpio.setup(13, gpio.IN, pull_up_down=gpio.PUD_OFF)
    gpio.add_event_detect(13, gpio.RISING)
    gpio.add_event_callback(13, callback)

    while True:
        ev.wait(1.0)
        if ev.is_set():
            break
    gpio.cleanup()
```

The following does seem to work:
```python

def proc1(ev):
    import RPi.GPIO as gpio    # delayed import

    def callback(channel):
        print("P11 triggered")

    gpio.setmode(gpio.BOARD)
    gpio.setup(11, gpio.IN, pull_up_down=gpio.PUD_OFF)
    gpio.add_event_detect(11, gpio.RISING)
    gpio.add_event_callback(11, callback)

    while True:
        ev.wait(1.0)
        if ev.is_set():
            break
    gpio.cleanup()

def proc2(ev):
    import RPi.GPIO as gpio    # delayed import

    def callback(channel):
        print("P13 triggered")

    gpio.setmode(gpio.BOARD)
    gpio.setup(13, gpio.IN, pull_up_down=gpio.PUD_OFF)
    gpio.add_event_detect(13, gpio.RISING)
    gpio.add_event_callback(13, callback)

    while True:
        ev.wait(1.0)
        if ev.is_set():
            break
    gpio.cleanup()
```