waveform80/rpi-lgpio #15: TypeError on cleanup()

johabeere, 2024-06-18:
## Outline: 
I migrated from RPi.GPIO because I need the add_event_detect functionality. Everything works as expected, except the PWM.stop() and GPIO.cleanup() functions: These cause the following stacktrace: 
```
Exception ignored in: <function PWM.__del__ at 0x7f58b01940>
Traceback (most recent call last):
  File "/home/phoenix/phoenix/software/venv/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 179, in __del__
  File "/home/phoenix/phoenix/software/venv/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 202, in stop
  File "/home/phoenix/phoenix/software/venv/lib/python3.11/site-packages/lgpio.py", line 1084, in tx_pwm
TypeError: unsupported operand type(s) for &: 'NoneType' and 'int
``` 
the last calling line is `return _u2i(_lgpio._tx_pwm(
      handle&0xffff, gpio, pwm_frequency, pwm_duty_cycle,
      pulse_offset, pulse_cycles))`, which leads me to believe that the "handle" object is set to None. I tried setting the gpiochip variable manually to see if maybe it had something to do with the software detecting the wrong GPIOChip, however the issue persists. I can't look into the `_lgpio._gpiochip_open() `method which creates the handle, but I can't seem to find the related module/file. 

## Information about my setup: 
Hardware: Raspberry Pi 4B 8GB
uname -a: `Linux firepi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux `
Python version: 3.11.2
Python is run inside a virtual environment created with venv. 
relevant pips: rpi-lgpio==0.6, lgpio==0.2.2.0

The error occurs every time a GPIO.PWM object has its __del__ ()/ stop() called, or the GPIO.cleanup() gets executed. 

Is this an upstream issue with lgpio? I wasn't able to find any similar error online. Since outputting to and reading from pins works, I assume that at least some part of the package is working. What can I do to debug this further? 

Best regards and thank you for your project!
Johannes

Just1nJ1, 2024-06-29:
Exactly the same issue here.

Just1nJ1, 2024-06-29:
# TL;DR
add `del p` before `GPIO.cleanup()`
i.e.
```
GPIO.setmode(GPIO.BOARD)
GPIO.setup(12, GPIO.OUT)
p = GPIO.PWM(12, 500)
p.start(0)
p.stop(0)
del p
GPIO.cleanup()
```
# Reason
After the program ends python will automatically delete all objects created during the program running, including the PWM object even if it is already 'stop', which requires the global variable `_chip` in GPIO. It's set to be None during the `GPIO.cleanup()` process, which is being used to do bitwise operation `&0xffff` and will cause the exception.

waveform80, 2024-07-28:
This could certainly be handled a bit more gracefully. Many thanks to @Just1nJ1 for the debugging; I'll see if I can clean this up a bit.

Butakus, 2024-10-30:
Hey, I am facing the same issue, and I wouldn't mind to work on a fix if you want some help.

I think checking if the _chip is None in the [PWM.stop method](https://github.com/waveform80/rpi-lgpio/blob/8de3453f743f9b1d3e290c3145dd52d9d4ed1298/RPi/GPIO/__init__.py#L193-L206) should be enough. I will send  a PR in a moment. Feel free to tell me if there is anything else I missed to fix it and update it.