waveform80/rpi-lgpio #16: Error when attempting to use GPIO2 or GPIO3

abishur, 2024-06-19:
I installed rpi-lgpio yesterday and setup a basic test program using GPIO2 and GPIO3 as inputs with the following basic program

```
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)
GPIO.setup(3, GPIO.IN, GPIO.PUD_DOWN)
```

and I get the following error message

```
 File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 659, in setup
    if _warnings and gpio in (2, 3) and pull in (PUD_UP, PUD_DOWN):
                                        ^^^^
NameError: name 'pull' is not defined
```

Now for best practices since pins 3 and 5 (GPIO2 and GPIO3) have hard wired 1K8 pull-ups to 3v3 for the i2c bus I should probably use different pins anyways since I want to pull it down, but I wanted to let you know regardless.

waveform80, 2024-07-28:
I was sure I'd fixed this one ... Hey [I did!](https://github.com/waveform80/rpi-lgpio/commit/a0f29d169a4e4495adc912b57108f29764f54476) -- but I note from the path in your report that you're using an apt installed version which suggests I need to get the current version into some distro somewhere -- can I ask which distro and which version of it you're using, eg RaspiOS bullseye?

DanielHabenicht, 2024-08-10:
@waveform80 got the same error: 

```
cat /proc/version
Linux version 6.6.31+rpt-rpi-v6 (serge@raspberrypi.com) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 Raspbian 1:6.6.31-1+rpt1 (2024-05-29)
```

waveform80, 2024-08-11:
Yes, it seems the version on RaspiOS is horribly out of date -- I'll see what I can do about that tomorrow (unfortunately I'm away from my desk this weekend)

DanielHabenicht, 2024-08-11:
Oh, no worries. Personal life is always more important than opensource! Can tell that from my own experience :)