waveform80/rpi-lgpio #17: Updated Pi5 bookworm today and get SOC error

quitmeyer, 2024-06-28:
This library is great and let's me use my code on pi5 or pi4 

normally i just:

sudo apt remove python3-rpi.gpio 

sudo apt install python3-rpi-lgpio 

and it works!

but i made a mistake today and updated the pi5, and now i get the SOC error that i got before. I tried reinstalling lgpio and uninstall gpio again, but nothing is working

Iphran, 2024-08-26:
Same Issue here but found out using:
apt show
the package version is: 0.4-1~rpt1, and rpi-lgpio already 0.6.
I just dump 0.6 file into where the package installed, and it's working.

mstumberger, 2024-11-10:
same issue here, it seems to me like a lack of permissions since it's working for me when running my code with sudo