waveform80/rpi-lgpio #18: Typo in setup() function argument

tmandys, 2024-07-06:
```
def setup(chanlist, direction, pull_up_down=PUD_OFF, initial=None):
....
-       if _warnings and gpio in (2, 3) and pull in (PUD_UP, PUD_DOWN):
+       if _warnings and gpio in (2, 3) and pull_up_down in (PUD_UP, PUD_DOWN):
 ```

LPS13, 2024-07-26:
I found this error as well. Trying to get the SafeShutdown.py script for my RetroFlag NESPi 4 case to function again after an OS upgrade.

waveform80, 2024-07-28:
Closing as a duplicate of #16, which is in fact already fixed (originally reported in #8) but which apparently still exists in some distro's packaged releases. Could I ask which distro and which version of that distro you're using? E.g. RaspiOS bullseye or something like that?

closed 2024-07-28 by waveform80

tmandys, 2024-11-04:
It was in RaspiOS bookworm.