waveform80/rpi-lgpio #19: more clarity on the purpose of this library in the documentation

AndySchroder, 2024-07-13:
I'm confused how this library relates to https://github.com/joan2937/lg . Why are there two python lgpio libraries? https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory mentions that lgpio can be used, but not rpi-lgpio. Why do we need rpi-lgpio at all if lgpio can be used directly by gpiozero?

Also, https://rpi-lgpio.readthedocs.io/en/latest/ suggests rpi-lgpio is for newer kernels. Is there any reason to still use RPi.GPIO? Are there some lacking features in rpi-lgpio because it is newer? https://rpi-lgpio.readthedocs.io/en/latest/differences.html describes some low level differences in behaviour, but it's not clear why to use rpi-lgpio at a high level.

waveform80, 2024-07-28:
I'll give a brief answer here and attempt to make this a bit clearer in the docs:

This library is purely a compatibility shim designed to make lgpio act like RPi.GPIO. In other words, it exists purely for those projects which still rely upon RPi.GPIO, but which can't use it because it's not supported on the Pi 5 (or on kernels which do away with the legacy sysfs gpio interface).

New projects should *not* be using this. They should either use lgpio directly, or perhaps libgpiod, or if they prefer something more high level, gpiozero (which sits on top of lgpio as well).

SakhileMamba, 2024-09-18:
> I'll give a brief answer here and attempt to make this a bit clearer in the docs:
>
> This library is purely a compatibility shim designed to make lgpio act like RPi.GPIO. In other words, it exists [...]

I tried using gpiozero in a .venv environment running a FAST API server to turn on an LED via HTTP requests from a Flutter frontend and it did not work.